CMU-SAFARI/ramulator #2: HBM IPC numbers

kkorgaon, 2015-06-12:
Hi,
 
When I run CPU trace simulation, I am seeing IPC numbers for HMC contrary to expectation. For some reason, these numbers do not follow the trend shown in the CAL paper. Attached a plot showing the results I am seeing. 
[image]

To debug the issue, I put this assert inside Memory.h header file. Turns out it never got called for HBM (as a note: it did got invoked for WideIO2)   

bool send(Request req)
{
        ... 
        // dispatch to the right channel
        assert(req.addr_vec[0] <= 0); 
        return ctrls[req.addr_vec[0]]->enqueue(req);
}

Can somebody provide pointers on what could be the issue? 

- Kunal

kevincha, 2015-06-26:
Hi Kunal,

Thanks for pointing out the issue. There was a problem on how the Ramulator maps which bits from the physical address to the channel/rank IDs. I've pushed a patch that fixes the problem to the main branch. So please check it out.

Regarding your assert, if the purpose is to make sure the function is called, shouldn't it be "assert(req.addr_vec[0] >= 0);",  instead of <= 0? req.addr_vec[0] is the channel ID, which is >= 0.

Thanks,
Kevin

closed 2015-06-26 by kevincha

kkorgaon, 2015-06-28:
Hi Kevin,

Thanks Kevin. To validate (for now) I ran this, and I do see a difference in IPC nos for HBM.  
./ramulator-cputrace cputraces/429.mcf;
      DDR3: 1.00000
      DDR4: 1.30493
      SALP: 1.22766
    LPDDR3: 0.71974
    LPDDR4: 0.93880
     GDDR5: 1.63193
       HBM: 1.86421
    WideIO: 0.95743
   WideIO2: 1.44096
     DSARP: 0.98984
    ALDRAM: 1.04933
    TLDRAM: 0.77555

The assert was for debugging purposes only. Now that the issue is fixed, it can be commented out.   

- Kunal