CMU-SAFARI/ramulator #4: Failure in installing Ramulator

bugubugu123, 2015-07-26:
Hi,
When I try to compile Ramulator, I met the following errors:

In file included from src/Main.cpp:1:
In file included from src/Processor.h:4:
In file included from src/Request.h:4:
In file included from /usr/include/c++/4.8/vector:62:
/usr/include/c++/4.8/bits/stl_construct.h:75:38: error: call to
      implicitly-deleted copy constructor of 'std::basic_ofstream<char>'
    { ::new(static_cast<void*>(__p)) _T1(std::forward<_Args>(__args)...); }
                                     ^   ~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/4.8/bits/stl_uninitialized.h:75:8: note: in instantiation of
      function template specialization
      'std::_Construct<std::basic_ofstream<char>, std::basic_ofstream<char> >'
      requested here
                std::_Construct(std::__addressof(*__cur), *__first);
                     ^
/usr/include/c++/4.8/bits/stl_uninitialized.h:117:2: note: in instantiation of
      function template specialization
      'std::__uninitialized_copy<false>::__uninit_copy<std::move_iterator<std::b                                             asic_ofstream<char>
      *>, std::basic_ofstream<char> *>' requested here
        __uninit_copy(__first, __last, __result);
        ^
/usr/include/c++/4.8/bits/stl_uninitialized.h:258:19: note: in instantiation of
      function template specialization
      'std::uninitialized_copy<std::move_iterator<std::basic_ofstream<char> *>,
      std::basic_ofstream<char> *>' requested here
    { return std::uninitialized_copy(__first, __last, __result); }
                  ^
/usr/include/c++/4.8/bits/stl_uninitialized.h:279:19: note: in instantiation of
      function template specialization
      'std::__uninitialized_copy_a<std::move_iterator<std::basic_ofstream<char>
      *>, std::basic_ofstream<char> *, std::basic_ofstream<char> >' requested
      here
      return std::__uninitialized_copy_a
                  ^
/usr/include/c++/4.8/bits/vector.tcc:413:15: note: in instantiation of function
      template specialization
      'std::__uninitialized_move_if_noexcept_a<std::basic_ofstream<char> *,
      std::basic_ofstream<char> *, std::allocator<std::basic_ofstream<char> > >'
      requested here
              = std::__uninitialized_move_if_noexcept_a
                     ^
/usr/include/c++/4.8/bits/vector.tcc:101:4: note: in instantiation of function
      template specialization 'std::vector<std::basic_ofstream<char>,
      std::allocator<std::basic_ofstream<char> >
      >::_M_emplace_back_aux<std::basic_string<char> >' requested here
          _M_emplace_back_aux(std::forward<_Args>(__args)...);
          ^
src/SpeedyController.h:67:33: note: in instantiation of function template
      specialization 'std::vector<std::basic_ofstream<char>,
      std::allocator<std::basic_ofstream<char> >
      >::emplace_back<std::basic_string<char> >' requested here
                cmd_trace_files.emplace_back(prefix + to_string(i) + suffix);
                                ^
src/Main.cpp:52:44: note: in instantiation of member function
      'ramulator::SpeedyController<ramulator::DDR3>::SpeedyController' requested
      here
        SpeedyController<DDR3>* ctrl = new SpeedyController<DDR3>(channel);
                                           ^
/usr/include/c++/4.8/fstream:599:28: note: copy constructor of
      'basic_ofstream<char, std::char_traits<char> >' is implicitly deleted
      because base class 'basic_ostream<char, std::char_traits<char> >' has a
      deleted copy constructor
    class basic_ofstream : public basic_ostream<_CharT,_Traits>
                           ^
/usr/include/c++/4.8/ostream:58:27: note: copy constructor of
      'basic_ostream<char, std::char_traits<char> >' is implicitly deleted
      because base class 'basic_ios<char, std::char_traits<char> >' has a
      deleted copy constructor
    class basic_ostream : virtual public basic_ios<_CharT, _Traits>
                          ^
/usr/include/c++/4.8/bits/basic_ios.h:66:23: note: copy constructor of
      'basic_ios<char, std::char_traits<char> >' is implicitly deleted because
      base class 'std::ios_base' has an inaccessible copy constructor
    class basic_ios : public ios_base
                      ^
1 error generated.
Makefile:14: recipe for target 'ramulator-dramtrace' failed
make: *** [ramulator-dramtrace] Error 1

My OS is openSUSE 13.2 (Harlequin) (x86_64), LLVM is 3.6.2  and gcc is 4.8.3...

Do you have any clue why these errors happen? Which environment is recommended?

Thanks a lot in advance.

SGwithADD, 2015-07-28:
I believe these are the same bugs that others have noticed when compiling with gcc 4.x (see #1).  @henryzh has posted a patch (#3) that works for gcc 5.1.  Unfortunately, there's a bug in gcc 4.8 that prevents it from properly compiling the lambda function, even with that patch.

Currently, we've seen some success with compiling using LLVM/clang (3.4+) on Ubuntu, using the version of the patch that's in the bugfix branch.  We're working to integrate the changes for gcc and LLVM to improve cross-platform compatibility.  We are hoping to have that done within the next week, but it's taking time for us to work through various system configurations to iron out all unexpected behavior.  Apologies for the delay.

bugubugu123, 2015-07-29:
Thanks a lot for letting me know the current status.

SGwithADD, 2015-07-29:
We have updated the code with some further changes for GCC 5.1 compatibility.  You can find these changes integrated in the master branch now.  Please let me know if this helps.  Thanks!

bugubugu123, 2015-08-03:
Thanks! I can compile it successfully with GCC5.2.

closed 2015-08-03 by bugubugu123