CMU-SAFARI/ramulator #6: Running GEM5+Ramulator with se.py+ruby with the new patch

dongha, 2015-08-07:
I could successfully compile GEM5+ramulator environment with SGwithADD's recent(yesterday) patch.
However, when I tried to run some benchmarks, it fails showing messages as follows.
______________
gem5.opt: build/ramulator/Ramulator/src/Config.cpp:12: void ramulator::Config::parse(const string &): Assertion `file.good() && "Bad config file"' failed.
gem5 Simulator System.  http://gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 compiled Aug  6 2015 15:03:19
gem5 started Aug  7 2015 11:38:22
gem5 executing on dongha-VirtualBox
<B>command line: ./build/X86_MESI_Two_Level/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=DDR3 -c mcf_base.amd64-m64-gcc43-nn -o ./inp.in
ramulator added</B>
Global frequency set at 1000000000000 ticks per second
Program aborted at tick 0
______________

I followed the README file which says "Run gem5 with --mem-type=ramulator and --ramulator-config=DDR3". 
But it seems ramulator fails to parse the '--ramulator-config=DDR3' option. 

So my questions are...
1) Do I have to modify se.py or other GEM5 files? Or am I missing something in the command line? If so, can you help me how?

And also, when I enable ruby with '--ruby' option, it seems gem5 does not include ramulator. 
2) Do I have to modify se.py to enable ramulator with ruby?

Anyway, I was lucky to find the patch as soon as I started using ramulator. Thanks.

SGwithADD, 2015-08-07:
Oh I see there's a typo in README.md.  The `--ramulator-config` argument takes in a filename, for a file that contains the configuration parameters of Ramulator.  We've included an example file, gem5-config.cfg, that contains a DDR3 configuration.  If you try `--ramulator-config=/path/to/ramulator/gem5-config.cfg`, does that work?

dongha, 2015-08-07:
works. thanks!

closed 2015-09-28 by i7mist

astromonk2013, 2016-01-21:
Could you please tell me if gem5-config.cfg still exists?Thanks

i7mist, 2016-01-21:
We now use a unified config file for all modes, i.e. there isn't a config file specifically for gem5 now, and you can use the config file in `configs` dir.

astromonk2013, 2016-01-21:
Thnaks, wanted to use the DDR3 config option.
Also getting a build error ramulator+gem5, any insights would be great:

build/ramulator/libramulator.a(Controller.os): In function `ramulator::Controller<ramulator::ALDRAM>::update_temp(ramulator::ALDRAM::Temp)':
build/ramulator/Ramulator/src/Controller.cpp:(.text+0x452): undefined reference to `ramulator::ALDRAM::aldram_timing(ramulator::ALDRAM::Temp)'
build/ramulator/libramulator.a(Controller.os): In function `ramulator::Controller<ramulator::TLDRAM>::issue_cmd(ramulator::TLDRAM::Command, std::vector<int, std::allocator<int> > const&)':
build/ramulator/Ramulator/src/Controller.cpp:(.text._ZN9ramulator10ControllerINS_6TLDRAMEE9issue_cmdENS1_7CommandERKSt6vectorIiSaIiEE[_ZN9ramulator10ControllerINS_6TLDRAMEE9issue_cmdENS1_7CommandERKSt6vectorIiSaIiEE]+0x1c0): undefined reference to `ramulator::TLDRAM::standard_name'
build/ramulator/Ramulator/src/Controller.cpp:(.text._ZN9ramulator10ControllerINS_6TLDRAMEE9issue_cmdENS1_7CommandERKSt6vectorIiSaIiEE[_ZN9ramulator10ControllerINS_6TLDRAMEE9issue_cmdENS1_7CommandERKSt6vectorIiSaIiEE]+0x1e0): undefined reference to `ramulator::TLDRAM::standard_name'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [build/X86/gem5.opt] Error 1
scons: building terminated because of errors.

i7mist, 2016-01-21:
It's duplicate question. Please see this thread: https://github.com/CMU-SAFARI/ramulator/issues/20

astromonk2013, 2016-01-21:
Thanks a lot!!

astromonk2013, 2016-01-21:
I tried to use the config files in configs like DDR3, HBM and TLDRAM , says bad config file
command line: ./build/X86/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=/home/ramulator-master/configs/HBM-config.cfg -c tests/test-progs/hello/bin/x86/linux/hello
Global frequency set at 1000000000000 ticks per second
gem5.opt: build/ramulator/Ramulator/src/Config.cpp:13: void ramulator::Config::parse(const string &): Assertion `file.good() && "Bad config file"' failed.
Program aborted at tick 0
Aborted (core dumped)
sobanerje@prodromou:~/gem5_mercurial/gem5-stable$ ./build/X86/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=/home/ramulator-master/configs/ALDRAM-config.cfg -c tests/test-progs/hello/bin/x86/linux/hello
gem5 Simulator System.  http://gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 compiled Jan 21 2016 12:16:24
gem5 started Jan 21 2016 13:49:44
gem5 executing on prodromou
command line: ./build/X86/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=/home/ramulator-master/configs/ALDRAM-config.cfg -c tests/test-progs/hello/bin/x86/linux/hello
Global frequency set at 1000000000000 ticks per second
gem5.opt: build/ramulator/Ramulator/src/Config.cpp:13: void ramulator::Config::parse(const string &): Assertion `file.good() && "Bad config file"' failed.
Program aborted at tick 0
Aborted (core dumped)
sobanerje@prodromou:~/gem5_mercurial/gem5-stable$ ./build/X86/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=/home/ramulator-master/configs/TLDRAM-config.cfg -c tests/test-progs/hello/bin/x86/linux/hello
gem5 Simulator System.  http://gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 compiled Jan 21 2016 12:16:24
gem5 started Jan 21 2016 13:49:52
gem5 executing on prodromou
command line: ./build/X86/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=/home/ramulator-master/configs/TLDRAM-config.cfg -c tests/test-progs/hello/bin/x86/linux/hello
Global frequency set at 1000000000000 ticks per second
gem5.opt: build/ramulator/Ramulator/src/Config.cpp:13: void ramulator::Config::parse(const string &): Assertion `file.good() && "Bad config file"' failed.
Program aborted at tick 0
Aborted (core dumped)

astromonk2013, 2016-01-21:
This is the command i am using, please advice
 ./build/X86/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=/home/ramulator-master/configs/ALDRAM-config.cfg -c tests/test-progs/hello/bin/x86/linux/hello

i7mist, 2016-01-22:
Hi, can you open another issue? The problem is not related with the original one

astromonk2013, 2016-01-22:
SUre, apologize.Will open another issue