CMU-SAFARI/ramulator #7: stoul invalid argument, ramulator terminated

sparsh0mittal, 2015-08-13:
I installed ramulator-cputrace and the cpu.trace runs fine, but when I run any file from cputraces folder, it shows the following error.

./ramulator-cputrace cputraces/481.wrf.gz 
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stoul
Aborted (core dumped)

Can you please help.

SGwithADD, 2015-08-18:
Have you tried unzipping the cpu traces and then running ramulator?

    gunzip cputraces/481.wrf.gz
    ./ramulator-cputrace cputraces/481.wrf

Ramulator does not support reading compressed trace files at the moment.

sparsh0mittal, 2015-08-20:
​I did that now, but I still see an error:

ramulator/cputraces$ ls
401.bzip2.gz  434.zeusmp.gz     444.namd.gz    456.hmmer.gz
464.h264ref.gz  *481.wrf.gz*
403.gcc.gz    435.gromacs.gz    445.gobmk.gz   458.sjeng.gz
470.lbm.gz      482.sphinx3.gz
429.mcf.gz    436.cactusADM.gz  447.dealII.gz  459.GemsFDTD.gz
 471.omnetpp.gz  483.xalancbmk.gz
433.milc.gz   437.leslie3d.gz   450.soplex.gz  462.libquantum.gz
 473.astar.gz

ramulator/cputraces$ gunzip 481.wrf.gz

ramulator/cputraces$ cd ..

ramulator$ ./ramulator-cputrace cputraces/481.wrf
terminate called after throwing an instance of 'std::out_of_range'
  what():  stoul
Aborted (core dumped)

ramulator$ ls cputraces/
401.bzip2.gz  434.zeusmp.gz     444.namd.gz    456.hmmer.gz
464.h264ref.gz  *481.wrf*
403.gcc.gz    435.gromacs.gz    445.gobmk.gz   458.sjeng.gz
470.lbm.gz      482.sphinx3.gz
429.mcf.gz    436.cactusADM.gz  447.dealII.gz  459.GemsFDTD.gz
 471.omnetpp.gz  483.xalancbmk.gz
433.milc.gz   437.leslie3d.gz   450.soplex.gz  462.libquantum.gz
 473.astar.gz

So, looks like issue is something else.

Thanks and Regards
Sparsh Mittal

i7mist, 2015-10-10:
Sorry for the delay to reply. Could you do the following things for us to spot the bug?
1. Could you pull the latest version and try to run `./ramulator --mode=cpu configs/DDR3-config.cfg 481.wrf` ?
2. Could you show a first few lines of the unzipped trace file? (481.wrf is okay)

Thanks
Tianshi

closed 2016-10-29 by SGwithADD