CMU-SAFARI/ramulator #11: Gem5-stable Building fails because of -Wno-undefined-bool-conversion (Unknown Warning

jeffreyflukman, 2015-11-13:
Hi Ramulator!

I have an issue when I tried to build the Gem5 Driven Section.

```
scons: Building targets ...
 [ISA DESC] ARM/arch/arm/isa/main.isa -> generated/inc.d
 [NEW DEPS] ARM/arch/arm/generated/inc.d -> arm-deps
 [ENVIRONS] arm-deps -> arm-environs
 [     CXX] ARM/sim/main.cc -> .o
error: unknown warning option '-Wno-undefined-bool-conversion'; did you mean '-Wno-bool-conversion'? [-Werror,-Wunknown-warning-option]
scons: *** [build/ARM/sim/main.o] Error 1
scons: building terminated because of errors.
```

I tried to locate this warning an I find it in gem5-0e86fac7254c-ramulator.patch line 37.
(It turns out that the error has a relation with my clang or g++ version, but I'm not sure whether it is right or not.)
** clang version : 3.4-1ubuntu3
** g++ version : Ubuntu 4.8.4-2ubuntu1~14.04

SGwithADD, 2015-11-13:
You're right - it's a version issue with your copy of clang/g++.  To run Ramulator in gem5 driven mode, you need clang 3.5+ or gcc 5.0+.  Does it work if you upgrade your version of clang?

jeffreyflukman, 2015-11-13:
I've upgraded my clang version to 3.5 and it could continue its compilation.
But now another error comes up : 

```
Error: can't find Python.h header in ['-I/usr/local/include/python2.7', '-I/usr/local/include/python2.7']
```

This error is strange for me, because when I tried to locate Python.h, it exists inside that directory.

Any other suggestion for this?

SGwithADD, 2015-11-18:
I'm afraid not.  I don't think we've encountered anything like that before.  I'm running Python 2.7.6 myself, with clang 3.5.0, on Ubuntu 14.04 LTS.  I take it you haven't had any luck since your last post?

jeffreyflukman, 2015-11-19:
I found the answer. So it turns out if we experienced an error before because we haven't installed the python-dev module or clang, we need to delete the whole build directory before we can re-compile the ramulator project again.

In my case it turns out that I haven't installed clang3.5+ in the beginning, but after I installed it, my ubuntu can't recognized it because it is installed with the clang++-3.5 name.

I create a symbolic link in /usr/local/bin yet it's still wouldn't work.

Then I found this link : http://comments.gmane.org/gmane.comp.emulators.m5.users/10463

Ali Saidi's respond there is the answer. :)

closed 2015-11-19 by jeffreyflukman

SGwithADD, 2015-11-19:
Ah yes.  I believe that's actually a gem5 artifact - it caches certain build parameters when you first create the build directory.  Anyhow, glad to see it's resolved.  Thanks for the update!