CMU-SAFARI/ramulator #12: ramulator addressing

mahzabeen, 2015-11-14:
Hi,
I have a very simple question about DRAM capacity provided in the Main.cpp. For example the DDR3 configuration says it is: DDR3_8Gb_x8 with one channel and one rank. Does it mean that the total DDR3 capacity here is 8GByte [(8Gb*8)/8] or is it only 8Gbit. From the address generation mechanism I found it to be 8 GByte because usually traces have byte level addressing and for this particular configuration we can use 33 bit for addressing (as I computed 33 =  6(ignore) + 8(col) + 3(bank) + 16(row)). But as the page size calculation suggests, each page size is 2 KByte (2^14 bits). But with 2^8 (256) addressable columns (each of which stores 64 bytes of data), page size will be 16 KByte (2^8 * 64 * 8 bits). 
Please let me know if I am misunderstanding any part. I am using the bugfix branch.

Thanks

i7mist, 2015-11-14:
Hi mahzabeen,

For DRAM size:
8Gbit is the capacity for one chip. As the channel width for DDR3 is 64, and the specification says x8, which means there are 8 lanes for this chip, so the total capacity for one DRAM with one rank should be 8Gbit * 64/8 = 64Gbit = 8 GByte, if the rank number doubles, the total capacity also doubles.

For page size:
It is 2KByte, the calculation method is column_number_in_one_bank * column_size (which is also channel_width), that is 2^8 * 64 bit = 16384 bit = 16Kb = 2 KB

Please let me know if you have other questions.

Also you can try the latest version with statistics, which has a "ramulator.dram_capacity" dumped in the statistics output file

Thanks,
Tianshi

mahzabeen, 2015-11-16:
Hi Tianshi,
Thank you so much for your prompt clarification. So for the previous example (DDR3_8Gb_x8) now I am calculating in the following way (please correct me if I am wrong):
Per bank capacity: 2^16 (rows) * 2^14 (page size in bits) = 2^30 bits
Per chip capacity: 8 (no. of banks) * 2^30 = 8 GBit
Per channel capacity: 8 (no. of chips) * 8 GBIt = 64 GBit = 8 GByte 

From the addressing mechanism it seems there is no bits to select the chip number explicitly and we activate a row to perform READ (64 bytes, since the addressing is done at 64 byte granularity). So my question is that, would this imply the opened row spans all the 8 chips? i.e. for each access we need to open 8 pages in 8 chips?

Also it will be great help if you please let me know that other than the user interface are there any internal difference between the master and bugfix branch?

I really appreciate your help.
Thanks
Mahzabeen

i7mist, 2015-11-20:
Your calculation for bank, chip, channel capacities are correct. And yes to your second question. Actually you can't index chip when access DRAM because the data of one column will be distributed among different chips. All chips will access the same column in the same row simultaneously in normal cases.

For the difference between bugfix and master branch: There are a lot of differences, mainly include a lot of statistics, bug fix and new interface. If you want a more detailed list of changes, please refer to commit log. Besides, bugfix is only a temporary branch. We'll push all stable features to master branch. So I suggest you to use master branch.

Thanks,
Tianshi

mahzabeen, 2015-11-20:
Hi Tianshi,
Thank you so much for your valuable feedback. I really appreciate your help.
-
Mahzabeen

closed 2015-11-20 by i7mist

SasinduWijeratne, 2020-06-07:
Hi Tianshi,
Where can I find a document regarding ramulator addressing?
Thanks,
Sasindu

sac2019, 2021-06-23:
> Hi mahzabeen,
> 
> For DRAM size:
> 8Gbit is the capacity for one chip. As the channel width for DDR3 is 64, and the specification says x8, which means there are 8 lanes for this chip, so the total capacity for one DRAM with one rank should be 8Gbit * 64/8 = 64Gbit = 8 GByte, if the rank number doubles, the total capacity also doubles.
> 
> For page size:
> It is 2KByte, the calculation method is column_number_in_one_bank * column_size (which is also channel_width), that is 2^8 * 64 bit = 16384 bit = 16Kb = 2 KB
> 
> Please let me know if you have other questions.
> 
> Also you can try the latest version in master branch with statistics, which has a "ramulator.dram_capacity" dumped in the statistics output file
> 
> Thanks,
> Tianshi

------------------------
Hi Tianshi @i7mist , 

Sorry, the way of calculating the page size is not clear for me? (or maybe I miss something!?).
you reach to a correct page size but not correct calculation !
what does "column_number_in_one_bank" mean?  it is not clear for me this number?!
 is it the same as "number of column address bits"? so why you consider it as 2*8 for DDR3-8G-8x?
--------------------------------------------------------------------------------------------------------------------------
 I propose this way to calculate the page size for any configuration of DDR3 or DDR4: 
the number of column address bits = 2^11 for DDR3-8G-x8 (based on datasheet or ddr3.h) 
and the page size can be calculating as follow:
Page size (in Byte) = 2^COLBITS × ORG/8, 
where COLBIT = the number of column address bits and ORG = the number of DQ bits (data queue =8 for DDR3-8G-8x).
so page size
for DDR3-8G-8x = 2^11 * 8 / 8= 2KB
for DDR3-8G-16x= 2^10 * 16/8 = 2KB  
for DDR3-gG-4x= 2^12 * 4 /8 = 2KB

you can check the above value with real one reported in page 2 of DDR3 datasheet:
 https://media-www.micron.com/-/media/client/global/documents/products/data-sheet/dram/ddr3/8gb_ddr3l.pdf?rev=92a422556160454e8e7cf0c70e706941

Regards,
Saeideh