CMU-SAFARI/ramulator #15: Problem with Ramulator.

greenmeent, 2016-01-06:
To use ramulator.
I followed README.md

---------------------------------------------------------------------------------------------------------
*Memory trace Driven*
$ cd ramulator
$ make -j
$ ./ramulator configs/DDR3-config.cfg --mode=dram dram.trace (here is problem)
---------------------------------------------------------------------------------------------------------

~/ramulator$ ls
configs    dram.trace                         LICENSE   plot.py    test_ddr3.py
cpu.trace  gem5-0e86fac7254c-ramulator.patch  Makefile  README.md  test_spec.py
cputraces  gem5-stable

~/ramulator$ ./ramulator configs/DDR3-config.cfg --mode=dram dram.trace
-bash: ./ramulator: No such file or directory

what should I do? Please help me~

i7mist, 2016-01-06:
Hi greenmeent,

I think your folder is incomplete. Could you try to clone it again? (It seems you even don't have src/ ...)

Thanks,
Tianshi

greenmeent, 2016-01-06:
thanks, Tianshi!!!!

I solve this problem but there is another problem....

./ramulator configs/DDR3-config.cfg --mode=cpu cpu.trace
ramulator: src/Config.cpp:43: void ramulator::Config::parse(const string &): Assertion `tokens.size() == 2 && "Only allow two tokens in one line"' failed.

What should I do? please

i7mist, 2016-01-06:
I think you may not use the latest version. Could you try to run "git pull" and make sure the last commit is "Update configs"?

Thanks,
Tianshi

closed 2016-10-29 by SGwithADD

zhizunbaowhy, 2022-07-04:
Hi,
I have the same problem.

./ramulator configs/DDR3-config.cfg --mode=cpu cpu.trace
ramulator: src/Config.cpp:43: void ramulator::Config::parse(const string &): Assertion `tokens.size() == 2 && "Only allow two tokens in one line"' failed.

I'm using the latest version. And I've already run "git pull". 
What should I do?
I would appreciate it if you could help me with the issue.