CMU-SAFARI/ramulator #18: Does cache really work?

bugubugu123, 2016-01-18:
Hi,
In the ramulator root, I ran the commands as follows:
>./ramulator configs/DDR3-A-config.cfg --mode=cpu --stats a.txt cputraces/483.xalancbmk
>./ramulator configs/DDR3-B-config.cfg --mode=cpu --stats b.txt cputraces/483.xalancbmk

The only difference between these two configuration files is that `cache` is set to `no` in DDR3-A-config.cfg while `l1l2` in DDR3-B-config.cfg. (DDR3-A-config is actually the default DDR3 config file coming along with the repository).
However, the two output files (`a.txt` and `b.txt`) are the same. So could you check whether the cache system is working or I have made some mistake in utilizing cache?

Thanks a lot in advance!

i7mist, 2016-01-18:
Hi bugubugu123,

The configuration is case sensitive so you should put exactly `L1L2` if you want the first two level cache model.

And the format of traces is different when you are using `L1L2` or `all` mode, i.e. it wasn't filtered by any cache model. It a trace is filtered by cache, any write request is due to cache eviction and always comes after a read request, otherwise read and write requests are independent. The format for unfiltered trace in cpu mode is similar to memory mode, with cpu instruction number at the beginning of each line. Our repository is still under heavily development so the README is not always up-to-date, sorry for the inconvenience.

Our traces have already been filtered by the first two level cache, only `L3` or `no` mode is applicable here. If you want to run multicore simulation by combining these traces, `L3` should be selected for cache and 'Random' for translation.

Thanks for your feedback! Please let me know if you have other questions.

Thanks,
Tianshi

bugubugu123, 2016-01-18:
Thanks a lot for your answer. Hope you can provide unfiltered traces in the future.

closed 2016-01-18 by i7mist