CMU-SAFARI/ramulator #20: Failed to compile with Gem5

ShawnLess, 2016-01-21:
I have cloned the Gem5 and reverted to the version 10231, and patched the gem5.

But when I try to compile the gem5, it also prompt  following error:
===================================================
 [    LINK]  -> ARM/gem5.opt
Undefined symbols for architecture x86_64:
  "ramulator::ALDRAM::aldram_timing(ramulator::ALDRAM::Temp)", referenced from:
      ramulator::Controller<ramulator::ALDRAM>::update_temp(ramulator::ALDRAM::Temp) in libramulator.a(Controller.os)
  "ramulator::TLDRAM::standard_name", referenced from:
      ramulator::Controller<ramulator::TLDRAM>::issue_cmd(ramulator::TLDRAM::Command, std::__1::vector<int, std::__1::allocator<int> > const&) in libramulator.a(Controller.os)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [build/ARM/gem5.opt] Error 1
scons: building terminated because of errors.
=====================================================
I have tried different architectures in building gem5, such as "scons build/x86/gem5.opt" and "scons build/ARM/gem5.opt", the same error occurs.

Any suggestion ?

i7mist, 2016-01-21:
Hi ShawnLess,

Try to add the following lines in `ext/ramulator/SConscript` after `DRAMFile('WideIO2.cpp')`
``` c++
DRAMFile('TLDRAM.cpp')
DRAMFile('ALDRAM.cpp')
```

I believe this can fix your problem.

BTW, I'm curious whether you did any modification to ramulator source code before compile it with gem5?

Thanks,
Tianshi

ShawnLess, 2016-01-22:
Great!  This fixed the problem pretty well.  Thanks.
I haven't done any modification to the source code. I just follow the instructions as on github.

commit 313a46b0 (2016-01-23): fix several issues related with gem5:

Fixes #20
fix missing definition compile error
Fixes #21
(gem5 mode) pass number of cpu to ramulator, specify cpuid in ramulator memory requests
commit d1d76efe (2020-12-13): fix several issues related with gem5:

Fixes #20
fix missing definition compile error
Fixes #21
(gem5 mode) pass number of cpu to ramulator, specify cpuid in ramulator memory requests
closed 2016-01-23 by i7mist