CMU-SAFARI/ramulator #21: Runtime error gem5+ramulator with x86 & DDR3 as config

astromonk2013, 2016-01-22:
command line: ./build/X86/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=/home/sobanerje/ramulator-master/configs/DDR3-config.cfg -c tests/test-progs/hello/bin/x86/linux/hello
Global frequency set at 1000000000000 ticks per second
gem5.opt: build/X86/base/statistics.hh:1026: void Stats::VectorBase<Stats::Vector, Stats::StatStor>::doInit(size_type) [Derived = Stats::Vector, Stor = Stats::StatStor]: Assertion `s > 0 && "size must be positive!"' failed.
Program aborted at tick 0
Aborted (core dumped)

Please advice
Thanks

i7mist, 2016-01-22:
Did you add any `VectorStat` statistics by yourself in ramulator? I guess this is because some `VectorStat` type statistics object doesn't call `init` function or use a non-positive value as parameter. (See how is `read_row_hits ` initialized in `src/Controller.h` if you don't know what is `init` function)

astromonk2013, 2016-01-22:
No I didnt change the vectorStat, I am just using the latest ramulator code.
Well I used a different gem5 config script inplace of se.py and looks like i am not getting that error anymore

gem5 compiled Jan 21 2016 12:16:24
gem5 started Jan 21 2016 14:18:12
gem5 executing on prodromou
command line: ./build/X86/gem5.opt ./configs/common/MemConfig.py --mem-type=ramulator --ramulator-config=/home/sobanerje/ramulator-master/configs/DDR3-config.cfg -c tests/test-progs/hello/bin/x86/linux/hello

But when I check m5out, the stats.txt generated doesnt have anything

ShawnLess, 2016-01-22:
Seem to be the same problem with me but with slightly different output :

shawnlessdeMacBook-Pro:gem5-stable shawnless$ ./build/X86/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=../ramulator/configs/DDR3-config.cfg -c tests/test-progs/hello/bin/x86/linux/hello

gem5 Simulator System.  http://gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 compiled Jan 21 2016 17:31:59
gem5 started Jan 22 2016 10:29:38
gem5 executing on shawnlessdeMacBook-Pro.local
command line: ./build/X86/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=../ramulator/configs/DDR3-config.cfg -c tests/test-progs/hello/bin/x86/linux/hello
Global frequency set at 1000000000000 ticks per second
Assertion failed: (s > 0 && "size must be positive!"), function doInit, file build/X86/base/statistics.hh, line 1026.
Program aborted at tick 0
Abort trap: 6

astromonk2013, 2016-01-22:
Only thing I changed while building gem5 with ramulator was C++ flag -Wno-undefined-bool-conversion to -Wno-bool-conversion as the build was failing as it seems GCC have any " -Wno-undefined-bool-conversion" flag option

astromonk2013, 2016-01-22:
removed assertion as a temporary solution, works now but ramulator stats in the stats file shows all 0..

i7mist, 2016-01-22:
I think I found the problem. It might be because some statistics added later to ramulator doesn't compatible with gem5 configuration. I'm trying to fix it now and will upload patch as soon as I make it.

astromonk2013, 2016-01-22:
Thank you.please update.we are waiting fr this issue to be resolved

i7mist, 2016-01-23:
I think I fixed this problem. please try the latest ramulator source code and gem5 patch and see whether the issue is solved. The problem is because we added some per-core statistics for ramulator multicore mode. Previously, the ramulator wasn't unaware of gem5 cpu number, so those statistics can't be successfully initialized when it is driven by gem5. I change both ramulator and gem5 to fix this issue so please update both ramulator and use the latest patch.

**For 0-value statistics**: Please run gem5 in a detailed cpu model (e.g. --cpu-type=detailed, timing etc.). **You need to explicitly specify that or it will use atomic mode as default mode**, which won't use detailed memory model for simulation speed, i.e. ramulator won't be called in that case.

I have tested the following command line and it works well on my machine:
single cpu:
```bash
./build/X86/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=ext/ramulator/Ramulator/configs/DDR3-config.cfg --cpu-type=detailed --caches -c tests/test-progs/hello/bin/x86/linux/hello
```
dual cpu:
```bash
./build/X86/gem5.opt ./configs/example/se.py --mem-type=ramulator --ramulator-config=ext/ramulator/Ramulator/configs/DDR3-config.cfg --cpu-type=detailed --caches --num-cpus=2 -c "tests/test-progs/hello/bin/x86/linux/hello;tests/test-progs/hello/bin/x86/linux/hello"
```

closed 2016-01-23 by i7mist

astromonk2013, 2016-01-25:
works!Thank you so much

commit 313a46b0 (2016-01-23): fix several issues related with gem5:

Fixes #20
fix missing definition compile error
Fixes #21
(gem5 mode) pass number of cpu to ramulator, specify cpuid in ramulator memory requests
commit d1d76efe (2020-12-13): fix several issues related with gem5:

Fixes #20
fix missing definition compile error
Fixes #21
(gem5 mode) pass number of cpu to ramulator, specify cpuid in ramulator memory requests