CMU-SAFARI/ramulator #22: x86 Gem5/Ramulator Segmentation Fault

Mappouras, 2016-01-22:
Hi all,

I have a problem trying to build gem5 with ramulator when compiling for x86 or ARM architecture. Curiously this problem does not occur for ALPHA architecture.

When building gem5 with ramulator for x86 I get the following error:

----------------------------------------
build/X86/arch/x86/process.cc:73:18: error: unused variable 'NumArgumentRegs' [-Werror,-Wunused-const-variable]
static const int NumArgumentRegs = sizeof(ArgumentReg) / sizeof(const int);
                 ^
build/X86/arch/x86/process.cc:81:18: error: unused variable 'NumArgumentRegs32' [-Werror,-Wunused-const-variable]
static const int NumArgumentRegs32 = sizeof(ArgumentReg) / sizeof(const int);
----------------------------------------

I order to resolve that I removed the -Wall flag. Then I am able to build gem5 but when I try to test it or run a full system simulation I get a "Segmentation Fault" error before the simulation even starts.

----------------------------------------
$ ./build/X86/gem5.prof configs/example/se.py -c tests/test-progs/hello/bin/x86/linux/hello
Segmentation fault (core dumped)
----------------------------------------

I am using clang 3.5, libstdc++, swig 3.0.2, gperftools are installed, I can build and run successfully gem5 and ramulator independently. These problems only arise when I try to include ramulator with gem5. Can anyone help?

Thanks,
George

i7mist, 2016-01-22:
I'll look into this issue. Could you tell me what OS you are using?

Mappouras, 2016-01-22:
3.19.0-25-generic Ubuntu

i7mist, 2016-01-23:
Hi, I figure that it is because ramulator requires clang or gcc-5 to compile, which will add -fomit-frame-pointer when -O3 is enabled. mcount unable to keep track of the calling graph when frame pointer is not available so the program crashes.

a workaround is to add '-fno-omit-frame-pointer' to both CCFLAGS and CXXFLAGS

Mappouras, 2016-01-23:
Thanks for your response i7mist. I tried that solution but it did not help. I still get the same warnings/errors that lead to a segmentation fault :(

i7mist, 2016-01-23:
I think it's not those warnings that lead to a segmentation fault. Those warnings are because some variables are not used when assert and debug print are not compiled (like in gem5.prof)

Could you give me more information? How did you add those flags and where did the segfault happen exactly? (can you use gdb to see the backtrace when the program crashes?) Besides, I suggest you set Werror=False in src/SConscript instead of removing Wall, so that you can still know what warning exists in your code. If all of them are about unused variable then just ignore them is fine.

Mappouras, 2016-01-23:
The problem is that I dont have direct access to the machine I am working on and gdb is not installed. I will ask the supervisor to install it for me but that may take a while.

All the warnings I get are the ones I mentioned above. I got them with the exact way you just described. 

I just tried to remove the -O3 flags to disable any optimizations and the problem seems to be resolved (I can run the hello world test...). However I am worried that the full system simulation will take for ever with out the optimizations.

The flags you mentioned I inserted them in the /path/to/gem5/SConstruct in lines 545 and 557 as follow
line 545: main.Append(CCFLAGS=['-fno-strict-aliasing','-fno-omit-frame-pointer']) 
line 557: main.Append(CXXFLAGS=['-std=c++11', '-stdlib=libstdc++','-fno-omit-frame-pointer'])

i7mist, 2016-01-23:
If you only worried about simulation speed, I think remove -pg can also solve this problem and you can remain -O3... gem5.perf and gem5.fast are compiled with -O3 without -pg

The way you added these flags seems to be correct... I don't have other ideas about what's happening temporarily... Another suggestion for you to test whether these flags will work is to write a very simple c++ program and compile it with clang++ -pg -O3 and clang++ -pg -O3 -fno-omit-frame-pointer respectively. See whether the former will cause a segfault and whether the latter will resolve it.

Mappouras, 2016-01-27:
Apparently I was not using the latest patch version for gem5. The latest updates (01/23/2016) seem to not have this issue

closed 2016-01-27 by Mappouras