CMU-SAFARI/ramulator #25: DRAM<T>:::update_timing() code

ShawnLess, 2016-01-27:
When I was inspecting code of DRAM.h, I was confused with DRAM.h:356 and DRAM.h:375. It says 

 __for (auto& t : timing[int(cmd)]) {__

but __timing__ is a private member of DRAM which is initialized in constructor with

__timing = spec->timing[int(level)];__

Where __spec->timing__ is a two dimensional data for all Spec.  So DRAM.timing is a vector and DRAM.timing[ int(cmd)] is a single structure, not a vector. 

Why the DRAM<T>::update_timing() uses the __for__ clause to traverse this scalar object ?

i7mist, 2016-01-27:
timing is a three dimensional data structure. For example, see the definition in `src/DDR3.h`:
```c++
     vector<TimingEntry> timing[int(Level::MAX)][int(Command::MAX)];
```
And you can read `init_timing` function in `src/DDR3.cpp` to see how this structure is instantiated

ShawnLess, 2016-01-28:
Oh, yes. Got it. Thanks. :+1:

closed 2016-01-28 by ShawnLess