CMU-SAFARI/ramulator #26: Gem5+Ramulator Error when running SPEC2006 benchmarks

Mappouras, 2016-02-03:
Hi all,

I came across a problem trying to test Gem5+Ramulator for SPEC2006.

I installed SPEC2006 and then I created static binaries for each benchmark. I tried to test them with gem5+ramulator but an error would pop up during their system emulation execution.

ERROR:
gem5.opt: build/X86/mem/request.hh:567: int Request::contectId() const: Assertion 'privateFlags.isSet(VALID_CONTEXT_ID)' failed

I run them like the example bellow:

build/X86/gem5.opt configs/example/se.py --cpu-type=detailed --caches --mem-type=ramulator --ramulator-config=/path_to_conf/DDR3-config.cfg -c /path_to_SPEC2006/staticbinaries/bzip2 -o /path_to_SPEC2006/input_for_bzip2

The emulations starts but it gets interrupted from the error mentioned above. Same problem for all the benchmarks I tried.

Simple programs like hello world execute just fine. This problem does not appear when I run gem5 without ramulator.

Any ideas how to fix that? Thanks

ChainedHope, 2016-02-11:
Having the same error with PARSEC benchmarking. 
Clang 3.6.2
GCC 5.2.1
Swig 2.0.12
Running on Ubuntu 15.10

Mappouras, 2016-02-11:
Sorry I did not give an earlier update. It seems that other memory modules (DramSim) have similar problems. You can go to the indicated file and change line 567 as follow:

if (privateFlags.isSet(VALID_CONTEXT_ID))
    return _contextId;
else
    return 0;

Of course that does not fix the problem, but it temporarily ignores the error.

i7mist, 2016-02-12:
Hi all, sorry for the delay. I have figured the reason and a patch to fix this with be pushed soon.

astromonk2013, 2016-03-02:
Hi i7mist, is there any update regarding the fix for this issue. Please let us know if there is a patch for this or what the fix is.Thanks

astromonk2013, 2016-03-02:
i believe the contextid in gem5 is performed just for the statistics. The ramulator is using the contextId for the source of the request or the coreid.Somehow the coreid or the context id is getting corrupted.I think its set to garbage value when the source of the request does not originate in the cpu but the cache. This might happen during a writeback. What confuses me is any access to the ramulator is a write back and it should fail immediately on a WR but it does not.

i7mist, 2016-03-03:
Hi astromonk2013, sorry for replying late. You are right about the source of corruption, it is happening during a writeback. I can fix this issue but I found there are still some other issues that make it corrupt when running some SPEC benchmarks. I didn't have time to fix that problem so I haven't pushed it to github. I'll try to get it done this week. Please stay tuned.

astromonk2013, 2016-03-03:
Thanks, just an update I am trying to use the threadId to pass the cpuid/coreId to the ramulator.lets see if that works.I'll look forward to your updates.

closed 2016-03-09 by i7mist

kvprathap, 2016-04-22:
Hello i7mist, I am at the latest commit(6ada583) and still get this error. I created a checkpoint with atomic cpu and restoring with '--mem-type=ramulator --ramulator-config=configs/LPDDR3-config.cfg --cpu-type=arm_detailed -r 1'.

gem5.opt: build/ARM/mem/request.hh:567: int Request::contextId() const: Assertion `privateFlags.isSet(VALID_CONTEXT_ID)' failed.

afairouz84, 2018-02-07:
Hi all,

Anyone fixed this issue?

I'm having the same error:

gem5.opt: build/X86/mem/request.hh:567: int Request::contextId() const: Assertion `privateFlags.isSet(VALID_CONTEXT_ID)' failed.

commit 6ada5837 (2016-03-09): fixes runtime error in Ramulator+Gem5

Fixes #26