CMU-SAFARI/ramulator #27: column address bit width

ShawnLess, 2016-02-04:
I'm a little confused by the column address bit width when I'm tracing the CMD of dram controller.

The column address bit width is subtracted by log2(prefetch_size) as at __src/Memory.h:122__;

__addr_bits[int(T::Level::MAX) - 1] -= calc_log2(spec->prefetch_size);__

This cutting causes incorrect cmd trace. For example, suppose  organization is 2Gbx8, channel with is 64. A single read command would return 8chip * 8bit * 8burst = 64 byte data. 

the address send to memory controller should start at 0 with 64 increment, that is:
__0x00000000__ 
__0x00000040__ 
__0x00000080__ 

the column address should start at 0 with 8 increment , that is 0x0, 0x8, 0x10. But actually  the cmd trace shows column address increases  with unit.
  __ACT          1:     0     0     0     0     0__
  __RD         12:     0     0     0     0     0__
  __RD         16:     0     0     0     0     1__
  __RD         20:     0     0     0     0     2__

I know that the low bits ( 3 bit for DDR3) is used only for burst order, not for addressing.  I'm wondering whether  the actually  address  send to dram is exactly as the trace shows. If it is,  please correct me if I misunderstand the column address of dram.

Thanks a lot in advance.

i7mist, 2016-02-09:
The last column address is not exactly column id, but is the index for cache line (for DDR3, it is 8 columns).

ShawnLess, 2016-02-10:
Confirmed that, thanks. :)

closed 2016-02-10 by ShawnLess