CMU-SAFARI/ramulator #30: New gem5 patch is generating build error

astromonk2013, 2016-03-09:
Please have a look at the build error:

build/X86/mem/ramulator.cc:37:19: error: no matching constructor for initialization of
      'ramulator::Gem5Wrapper'
    wrapper = new ramulator::Gem5Wrapper(configs, system()->cacheLineSize());
                  ^                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
build/ramulator/Ramulator/src/Gem5Wrapper.h:20:5: note: candidate constructor not viable: no known
      conversion from 'ramulator::Config' to 'const string' (aka 'const basic_string<char>') for 1st
      argument
    Gem5Wrapper(const string& config_file, int cacheline);
    ^
build/ramulator/Ramulator/src/Gem5Wrapper.h:14:7: note: candidate constructor
      (the implicit copy constructor) not viable: requires 1 argument, but 2 were provided
class Gem5Wrapper

Thanks

i7mist, 2016-03-11:
It seems you are not using the latest version of ramulator source code? Since we changed the interface after this commit: https://github.com/CMU-SAFARI/ramulator/commit/415d8f7f51e6397f4bf5678dd32d78226c81b8fd

Could you double check whether you are at the correct branch?

astromonk2013, 2016-03-11:
I can find the changes you made to Gem5wrapper, controller, memory in commit https://github.com/CMU-SAFARI/ramulator/commit/415d8f7f51e6397f4bf5678dd32d78226c81b8fd
as well as I updated after you made the changes after that commit.

But you could close this issue as I manually made the changes in the latest commit you made and it works.

Btw  I would request some info regarding the cache model you have in ramulator .should I open another issue?

astromonk2013, 2016-03-11:
and you are right I was using the older ramulator copy I have so please close this issue.
Sorry about that

closed 2016-03-11 by astromonk2013