CMU-SAFARI/ramulator #47: timing lookup table typos?

stevekulick, 2018-01-17:
DDR4.h (DDR3.h as well)
    // CAS <-> CAS (between sibling ranks)
    t[int(Command::RD)].push_back({Command::RD, 1, s.nBL + s.nRTRS, true});
    t[int(Command::RD)].push_back({Command::RDA, 1, s.nBL + s.nRTRS, true});
    t[int(Command::RDA)].push_back({Command::RD, 1, s.nBL + s.nRTRS, true});
    t[int(Command::RDA)].push_back({Command::RDA, 1, s.nBL + s.nRTRS, true});

    //ssk are these typos? looks like they should be wr > wr
    t[int(Command::RD)].push_back({Command::WR, 1, s.nBL + s.nRTRS, true});
    t[int(Command::RD)].push_back({Command::WRA, 1, s.nBL + s.nRTRS, true});
    t[int(Command::RDA)].push_back({Command::WR, 1, s.nBL + s.nRTRS, true});
    t[int(Command::RDA)].push_back({Command::WRA, 1, s.nBL + s.nRTRS, true});

    // ssk these are the correct eqns for rd > wr
    t[int(Command::RD)].push_back({Command::WR, 1, s.nCL + s.nBL + s.nRTRS - s.nCWL, true});
    t[int(Command::RD)].push_back({Command::WRA, 1, s.nCL + s.nBL + s.nRTRS - s.nCWL, true});
    t[int(Command::RDA)].push_back({Command::WR, 1, s.nCL + s.nBL + s.nRTRS - s.nCWL, true});
    t[int(Command::RDA)].push_back({Command::WRA, 1, s.nCL + s.nBL + s.nRTRS - s.nCWL, true});

arthasSin, 2018-01-17:
I think you meant the .cpp files DDR4.cpp and DDR3.cpp. Yes, seems like you are right about the typos. It should definitely be WR-to-WR timing as RD-to-WR timing is defined later. We will get it fixed. Thanks for the heads up.

hmhmhey, 2018-02-13:
It seems there's another typo in DDR4.cpp. 

`    /*** Bank Group ***/ 
    t = timing[int(Level::BankGroup)];
    // CAS <-> CAS
    t[int(Command::RD)].push_back({Command::RD, 1, s.nCCDL});
    t[int(Command::RD)].push_back({Command::RDA, 1, s.nCCDL});
    t[int(Command::RDA)].push_back({Command::RD, 1, s.nCCDL});
    t[int(Command::RDA)].push_back({Command::RDA, 1, s.nCCDL});

    t[int(Command::WR)].push_back({Command::WR, 1, s.nCCDL});
    t[int(Command::WR)].push_back({Command::WRA, 1, s.nCCDL});
    t[int(Command::WRA)].push_back({Command::WR, 1, s.nCCDL});
    t[int(Command::WRA)].push_back({Command::WRA, 1, s.nCCDL});

    // looks like this part should define RD > WR
    t[int(Command::WR)].push_back({Command::WR, 1, s.nCCDL});
    t[int(Command::WR)].push_back({Command::WRA, 1, s.nCCDL});
    t[int(Command::WRA)].push_back({Command::WR, 1, s.nCCDL});
    t[int(Command::WRA)].push_back({Command::WRA, 1, s.nCCDL});

    t[int(Command::WR)].push_back({Command::RD, 1, s.nCWL + s.nBL + s.nWTRL});
    t[int(Command::WR)].push_back({Command::RDA, 1, s.nCWL + s.nBL + s.nWTRL});
    t[int(Command::WRA)].push_back({Command::RD, 1, s.nCWL + s.nBL + s.nWTRL});
    t[int(Command::WRA)].push_back({Command::RDA, 1, s.nCWL + s.nBL + s.nWTRL});
`

I'd appreciate it if you could check whether the following code is correct.
`    /*** Bank Group ***/ 
    t = timing[int(Level::BankGroup)];
    // CAS <-> CAS
    t[int(Command::RD)].push_back({Command::RD, 1, s.nCCDL});
    t[int(Command::RD)].push_back({Command::RDA, 1, s.nCCDL});
    t[int(Command::RDA)].push_back({Command::RD, 1, s.nCCDL});
    t[int(Command::RDA)].push_back({Command::RDA, 1, s.nCCDL});

    t[int(Command::WR)].push_back({Command::WR, 1, s.nCCDL});
    t[int(Command::WR)].push_back({Command::WRA, 1, s.nCCDL});
    t[int(Command::WRA)].push_back({Command::WR, 1, s.nCCDL});
    t[int(Command::WRA)].push_back({Command::WRA, 1, s.nCCDL});

    // this part!   
    t[int(Command::RD)].push_back({Command::WR, 1, s.nCL + s.nCCDL + 2 - s.nCWL});
    t[int(Command::RD)].push_back({Command::WRA, 1, s.nCL + s.nCCDL + 2 - s.nCWL});
    t[int(Command::RDA)].push_back({Command::WR, 1, s.nCL + s.nCCDL + 2 - s.nCWL});
    t[int(Command::RDA)].push_back({Command::WRA, 1, s.nCL + s.nCCDL + 2 - s.nCWL});

    t[int(Command::WR)].push_back({Command::RD, 1, s.nCWL + s.nBL + s.nWTRL});
    t[int(Command::WR)].push_back({Command::RDA, 1, s.nCWL + s.nBL + s.nWTRL});
    t[int(Command::WRA)].push_back({Command::RD, 1, s.nCWL + s.nBL + s.nWTRL});
    t[int(Command::WRA)].push_back({Command::RDA, 1, s.nCWL + s.nBL + s.nWTRL});
`

arthasSin, 2018-02-13:
@hmhmhey yes, we are aware of that issue. I will go over all timing parameters soon to make sure everything is in order. Your changes make sense but I am not sure if it is entirely correct. I will need to check the DDR4 specification. Thanks.

RSpliet, 2018-07-12:
@hmhmhey Your proposed change is incorrect. Instead, the RD-WR timing at the bank group level can simply be scrapped. The correct timing constraints are defined at the rank level, and should be independent of whether you hit the same or a different bank group (because tCL is trivially larger than both tCCDS and tCCDL). At this level, it could be that tCCDS needs to be replaced with tBL; in practice you'll find that the two are always equal for DDR4. If they weren't, you'd never under any circumstance be able to achieve 100% data bus utilisation, in which case raising the clock to those levels makes no sense any more.

I have opened a pull request today that corrects several timing issues in DDR4, one of which corrects situations under which timing violations could occur. I shall correct the rank-level timing constraints for RD->WR in a follow-up patch for maintainability reasons and include it in the pull request. This will not affect the correctness of ramulator's results.

RSpliet, 2018-07-13:
Pull request #57 has just been merged, meaning @hmhmhey's issue has been resolved. As for the original issue posted by @stevekulick , my current experimental set-up has only a single rank. As such, I am unable to test or verify any changes in that area.