CMU-SAFARI/ramulator #48: New ramulator running forever in cpu mode

mahzabeen, 2018-01-23:
Hi,
I was trying out the most recent ramulator downloaded today (1/23/2018), in cpu mode it is running forever with the simple cpu.trace that comes with ramulator.

Any help on this will be greatly appreciated.

Thanks

arthasSin, 2018-01-23:
Is it possible for you to share with us the config file that you are using? We would like to try to reproduce the issue. Please note that although the cpu.trace file contains just a few instructions, those instructions will be repeatedly executed until the expected limit for the number of instructions defined in the configuration file is reached. This is the default behavior and to change that you need to make minor changes in Processor.cpp. I can tell you about it more if this is the problem you face. Can you try lowering the expected_limit_insts value in the config file and make sure that the run compeletes quickly?

mahzabeen, 2018-01-24:
Hi,
Thank you so much for the feedback. I am attaching the configuration file I was running. I have also tried with expected_limit_insts = 1000 but it did not help.
Can you please let me know what changes I need to make in Processor.cpp so that the program runs as long there are requests in the trace file, i.e. it only depends on the trace file not on any expected limit.
Thank you once again.

mahzabeen, 2018-01-24:
Here is the DDR3-config.cfg file that I have used:
########################
# Example config file
# Comments start with #
# There are restrictions for valid channel/rank numbers
 standard = DDR3
 channels = 1
 ranks = 1
 speed = DDR3_1600K
 org = DDR3_2Gb_x8
# record_cmd_trace: (default is off): on, off
 record_cmd_trace = off
# print_cmd_trace: (default is off): on, off
 print_cmd_trace = off

### Below are parameters only for CPU trace
 cpu_tick = 4
 mem_tick = 1
### Below are parameters only for multicore mode
# When early_exit is on, all cores will be terminated when the earliest one finishes.
 early_exit = on
# early_exit = on, off (default value is on)
# If expected_limit_insts is set, some per-core statistics will be recorded when this limit (or the end of the whole trace if it's shorter than specified limit) is reached. The simulation won't stop and will roll back automatically until the last one reaches the limit.
 expected_limit_insts = 200000000
 warmup_insts = 100000000
 cache = no
# cache = no, L1L2, L3, all (default value is no)
 translation = None
# translation = None, Random (default value is None)
#
########################

arthasSin, 2018-01-25:
Please pull the latest changes. The trace file will not be executed over and over again if you make 'expected_limit_insts=0' in your config file. When 'expected_limit_insts; is larger than 0, the input trace file is repeatedly executed until the core executes at least 'expected_limit_insts' instructions.

Also, be careful with the number of the warm-up instructions (i.e., 'warmup_insts'). You should either set it to 0 to skip the warmup period entirely or set it to a number smaller than the number of instructions in the input trace file. Statistics for the instructions executed during the warmup period are discarded.