CMU-SAFARI/ramulator #51: Ramulator does not follow the order of the trace file in DRAM mode

shashankshashank44, 2018-04-05:
Hi,

When I run ramulator in DRAM mode, the order of Issue commands are not in the order given by the trace file file.
for example, the order of trace file is
Address_1 W
Address_1 R
Address_2 R
Address_3 R
the order of issue commands are
Address_1 R
Address_2 R
Address_3 R
Address_1 W

The "Write" should be addressed first. Don't the simulator follow the order in DRAM mode?

arthasSin, 2018-04-05:
Hi, 

The default request scheduling policy (i.e., FRFCFS) in Ramulator reorders the requests to achieve better performance. Depending on the size of the read/write request queues, the scheduler is able to choose a request to schedule based on the priorities implemented in the scheduling policy. For example, FRFCFS prioritizes requests that would result in a row-buffer hit. 

Besides the scheduling policy, another factor that would reorder the requests is the fact that the memory controller services only read requests for a while, and then switches to servicing only write requests. The memory controller does not interleave the two request types in a fine-grained manner. The reason for that is reducing the overhead of DRAM bus turnaround when switching between read and write modes on the DRAM data bus.

To force in-order request scheduling, you may try changing the read/write queue sizes to 1. However, you may still see some reordering as the memory controller will try to group the read and write requests to perform fewer DRAM bus turnarounds. To properly force in-order scheduling, you may want to modify the memory controller logic to unify the read and write request queues.

shashankshashank44, 2018-04-05:
Thanks a lot for the reply,
I have a better idea now.

closed 2018-04-05 by shashankshashank44