CMU-SAFARI/ramulator #52: Cache::send silently fails when it goes to lower levels?

johnjohnlin, 2018-05-05:
Hello, I first want to thank to the authors for making such useful project. I am using the DRAM part of Ramulator standalone (i.e. not using the main file and Processor.cpp; no GEM5), and everything goes fine until I use a shared cache. In `Cache::send`, even if `lower_cache` fails to send the request, this function still returns true while `req` is lost forever (it is not pushed to any `wait_list`). Specifically, the code segment is:
```cpp
if (!is_last_level) {
  lower_cache->send(req);
} else {
  cachesys->wait_list.push_back(
      make_pair(cachesys->clk + latency[int(level)], req));
}
return true;
```

I try to reproduce this problem with a minimal example. In this example, I use 2 L1 and 1 shared L2 cache(s), sending read request 0~9 for both cores.

```cpp
shared_ptr<CacheSystem> csys(new CacheSystem(configs, mem_send));
csys->first_level = Cache::Level::L1;
csys->last_level = Cache::Level::L2;
// Cache: direct map, line size = 1, size = 64
const int MSHR = 2;
const int SIZE = 64;
Cache *priv0 = new Cache(SIZE, 1, 1, MSHR, Cache::Level::L1, csys);
Cache *priv1 = new Cache(SIZE, 1, 1, MSHR, Cache::Level::L1, csys);
Cache *share = new Cache(SIZE, 1, 1, 2*MSHR, Cache::Level::L2, csys);
priv0->concatlower(share);
priv1->concatlower(share);
for (int i = 0, j0 = 0, j1 = 0; i < 1000; ++i) {
    if (j0 < 10) {
        if (priv0->send(Request(j0, Request::Type::READ, [share](Request r) {share->callback(r);}))) {
            j0++;
        }
    }
    if (j1 < 10) {
        if (priv1->send(Request(j1, Request::Type::READ, [share](Request r) {share->callback(r);}))) {
            j1++;
        }
    }
    csys->tick();
    mem->tick();
}
```
During the first call of `priv1->send(...)`, since core 0 has locked the only line (since L1, L2 are direct map), the underlying `lower_cache->send(req);` fails at `cache_set_unavailable++; return false;`. However I don't have a simple way to know this happens.

So my problem is: is this a correct behavior; or am I doing something wrong? The full listing is attached as [my_main.txt](https://github.com/CMU-SAFARI/ramulator/files/1976434/my_main.txt) and can be compiled by (g++7.3.1; executed after a standard Ramulator compilation):
```sh
g++ my_main.cpp obj/*.o -DRAMULATOR -g
```

arthasSin, 2018-05-14:
Hi,

Seems like there is a problem with recursive calls to the `Cache::send` function. How did you find out this problem? Did the simulation enter an infinite loop?

Could you try changing 
```
if (!is_last_level) {
  lower_cache->send(req);
} else {
  cachesys->wait_list.push_back(
      make_pair(cachesys->clk + latency[int(level)], req));
}
return true;
```

to 

```
if (!is_last_level) {
  return lower_cache->send(req);
} else {
  cachesys->wait_list.push_back(
      make_pair(cachesys->clk + latency[int(level)], req));
}
return true;
```

johnjohnlin, 2018-05-15:
Yes, I find this because the simulation enter a infinite loop.
An address is put to a queue since this function return true, but it is never popped since the request vanishes.
I try the modified version and it works in my own case.
Also, I test on one command to check whether this modification break the original code.
It works and I create a PR about it. https://github.com/CMU-SAFARI/ramulator/pull/53

```bash
./ramulator configs/DDR3-config.cfg --mode=cpu cputraces/401.bzip2 cputraces/401.bzip2
```

However, I somewhat wonder that, at a few lines above, this function allocate some resources,
and will this cause any problem if we return a false value here?

```cpp
auto newline = allocate_line(lines, req.addr);
```

arthasSin, 2018-05-15:
Seems like you are right. Allocating a cache line in the first level cache, while `send()` returns `false`, incorrectly changes the state of the cache. Although it may not result in functionally incorrect operation, the cache may not be simulated properly in terms of timing. I will think about a better way of fixing this issue. Thanks for sharing the code for reproducing this issue.

arthasSin, 2018-05-18:
I just pushed a fix for this issue. Hopefully, it resolves your problem.

johnjohnlin, 2018-05-21:
Thanks for the fast update. This version works smoothly so far.

closed 2018-05-21 by johnjohnlin