CMU-SAFARI/ramulator #54: read request order and callback order

timesir, 2018-06-12:
why isn't the tead request order  the same with callback order in ramulator? This is not the same with real memory access behavior. How to make it ordered?

arthasSin, 2018-06-13:
I am not sure if I fully understood your problem. Please provide more information if you are not satisfied with the following:

The memory controller services memory requests based on a request scheduling policy. The default scheduling policy for Ramulator is first-ready-first-come-first-serve (FRFCFS), which prioritizes requests that hit in the row-buffer of the bank that the request is accessing. The request scheduling priority affects the order in which the requests are serviced. 

However, that is not the only reason why the order of requests submitted to the request queue is different from the order of completion of read requests, i.e., "callback order". Even if you try with first-come-first-serve (FCFS) policy, you will still observe reordering due to variation in servicing requests targeting different banks. For example, let's assume read request A is older than read request B in the request queue. With FCFS policy, the memory controller first services A, which targets bank 0. If bank 0 has an open row that is different than the row A targets, the memory controller will precharge bank 0. While precharging, the memory controller will utilize the "free" time to service B, which targets bank 1. If bank 1 already has request B's row opened or if bank 1 is already precharged, the data of request B will be read from memory sooner than A's data.  

Can you elaborate more on what you mean by "This is not the same with real memory access behavior"?

Cassiel-girl, 2018-06-13:
I encountered the same problem very confusing. When accessing memory through AXI, the first read request will first delivery data to master, but ramulator delivery data out of order. It seems to  memory controller configure parameter  "strict order".

timesir, 2018-06-14:
Thank all you for reply. I want to express the same as cassiel said.