CMU-SAFARI/ramulator #55: what'

Cassiel-girl, 2018-06-18:


arthasSin, 2018-07-02:
What do you mean by length? A read request typically causes a cache line (64 bytes) to be fetched from the DRAM. Is it what you are asking?

Cassiel-girl, 2018-07-05:
Yes, but why not 64bit? After all, data bus is 64bit.

arthasSin, 2018-07-05:
Cache line size typically is 64 bytes. The DRAM data bus could be 64-bits wide but a single READ or WRITE command performs 64-bytes of transfer over multiple consecutive cycles to serve a last-level cache request with a single DRAM command. For example, by setting the _burst length_ to 8, eight times 64-bit data is transferred to fill or write-back a 64-byte cache line.

Cassiel-girl, 2018-07-14:
I can't find the source code where means "eight times 64-bit data is transferred to fill". Could you tell me where in the code?

mfilteau, 2019-05-24:
It would be really nice if the actual bus width and burst size would be configurable.