CMU-SAFARI/ramulator #65: Burst mode

marcodamico, 2019-02-27:
I have a question regarding burst mode. Is it considered in ramulator?
I see the prefetch size of the DDR configuration is used to calculate tx.
In src/Memory.h:118 you define:
int tx = (spec->prefetch_size * spec->channel_width / 8);
I don't really understand why you divide by 8.
E.g. with a burst of 8 and a channel of 64bits, the DRAM should read 64 * 8 bits of data and send them through the bus in multiple clock cycles.
It is possible to have a Read followed by a write happening during the burst transfer, would this case be supported by your emulator?

arthasSin, 2019-02-27:
The division by 8 in src/Memory.h:118 is to calculate tx as a number of bytes.  In Ramulator, prefetch_size is the burst length. It is set in the header file of your target specification, e.g., DDR4.h

marcodamico, 2019-02-28:
Thank you for your answer. In the last question, I am wondering if the minimum READ-to-WRITE latency is respected, to avoid conflicts on the data bus.

arthasSin, 2019-02-28:
yes, it is. You will find all timing constraints in the .cpp files of the standards (e.g., DDR4.cpp)

closed 2019-04-16 by marcodamico