CMU-SAFARI/ramulator #67: On the fly burst mode

marcodamico, 2019-03-29:
Hello,
some DRAM models allow changing the burst length on the fly. Would be difficult to implement it in ramulator? What need to be changed every time a new burst length is set?

Thank you.

RSpliet, 2019-05-11:
Do you have any DRAM technology in mind? In the DDR family, I believe this feature was removed since DDR3. Previous generations of DDR SDRAM seem unsupported in Ramulator. Likewise, the JEDEC GDDR5 spec defines a fixed burst length of 8 beats, while earlier versions aren't supported in Ramulator.

* DDR3 and DDR4 do support "burst chop" to cut bursts short. However, this only impacts power consumption, not timing.

marcodamico, 2019-05-13:
According to JEDEC standard JESD209-4, LPDDR4 accepts on the fly burst mode.

RSpliet, 2019-05-13:
Thanks. I hadn't looked in the LPDDR4 specs previously, that's an interesting find.

I _think_ that on for regular DDR technology this feature was dropped because in practice all transfers are cache-line width. There's no on-the-fly decision to be made and the command rate is high enough to accomodate longer bursts of data simply by issuing commands back-to-back. The same might be true for tablets and mobile phones using LPDDR, but for more embedded devices there might be some use-cases for on-the-fly burst length.

Which brings me to your question of "difficulty in implementing". On what level of the model do you want to support this? Judging by comments in source code, Ramulator currently assumes burst lengths are 16 beats on LPDDR4. So to add support for 32-beat bursts without changing any interfaces, you could simply add two new commands RD2BL/WR2BL or something like that (I'll leave the bike-shedding over names up to others) and add all the proper timing constraints for these two commands in LPDDR4.h. That should be sufficient if you're the one designing a new memory controller. Since Ramulator doesn't model MR updates, the same mechanism would be valid for fixed burst-lengths (of either 16 or 32 for LPDDR4). You can now interface with Ramulators DRAM object and make it validate timing for you.
However, if you need support higher up the hierarchy (Ramulator's memory controller or cache hierarchy), we need to acknowledge that such "long reads" are not part of every standard. You'll have to think of the constraints for which you choose to issue short vs. long bursts, and create new policies for that. Simply hacking up the existing cache and memory controller probably won't do it, clean design will more likely involve creating some new objects, using inheritance for partial code re-use, and somehow requesting from the DRAM system whether on-the-fly bursts are even supported.

marcodamico, 2019-05-16:
Thanks for your answer. Regarding LPDDR4.h code, I guess it is enough updating the entries containing "nBL" value in timing table, correct?