CMU-SAFARI/ramulator #75: How to generate cpu.trace like the supported spec2006?

john-Ly, 2019-08-24:
Hi, there.
I'd like to use other benchmarks(like pbbs, parsec etc.), can you share how to generate the cpu.trace in detail?
Thx.

wwwenh, 2023-09-05:
I had a similar issue as well. Have you managed to resolve it? If so, could you please share the solution? It would be greatly appreciated.

Thank you!

nisabostanci, 2023-09-29:
Hi john-Ly and wwwenh,

You can generate traces for different applications with [our trace generator tool](https://github.com/CMU-SAFARI/ramulator/tree/master/trace_generator). After following the instructions to build the tool, you can use a command similar to the one below to generate CPU traces:

`./tracegenerator.sh -mode cpu [other options] -- [command to run the selected application]`

Please see the README file in the directory for other options you can use.