CMU-SAFARI/ramulator #89: request.depart set to 0 when created

hanm2019, 2020-11-22:
I found the parameter depart in Request.h is not init to any value. I think it may cause some problem when a WRITE request send to the Memory and in the callback function, the request.depart is a random value, since this parameter hasn't been set to any value. Maybe set this value to 0 at the init time is better.

arthasSin, 2020-11-23:
Request.depart is only used for READ requests and depart is set right before a READ req is inserted into the pending queue. So, having depart uninitialized when creating a Request object shouldn't cause a problem but it is a good practice to set initial values for all member variables of course. Thanks for the feedback

hanm2019, 2020-11-23:
since the issue[40](https://github.com/CMU-SAFARI/ramulator/issues/40), the WRITE req also call the callback, so if the callback function don't make sure the operation type, and do `r.depart -  r.arrive` directly, it may cause some problem.

in the Main.cpp run_dramtrace() 
```
auto read_complete = [&latencies](Request& r){latencies[r.depart - r.arrive]++;};
```
 maybe get a wrong result.