CMU-SAFARI/ramulator #103: gem5 version 10231 now is unavailable

asd055189, 2021-09-01:
Hello, I'm try to build a gem5-driven ramulator, the site `http://repo.gem5.org/gem5-stable`  given by the README is down.
Also, I can't find the working version of gem5 with ramulator.
Does anyone have any idea?

nidsam, 2022-02-02:
@asd055189 I am facing same issue. did you find solution?

asd055189, 2022-02-02:
@nidsam 
checkout this version：）
https://gem5.googlesource.com/public/gem5/+/0be64ffe2f4ff8824b3084362706ffbf456ea490

nidsam, 2022-02-02:
@asd055189 thanks for the reply :) 
can you please tell me what commands you used with this link?
I am not able to do it.

asd055189, 2022-02-02:
@nidsam
you can download it simply by git clone

Also remind the dependency of your scons and python version.

I choose Ubuntu18.04LTS on WSL2 which has higher dependency.