CMU-SAFARI/ramulator #109: stream read & write requests are handled every 4 cycles?

Miayiyi, 2021-12-22:
Hi, 

I tried running stream read/write traces with DDR3 only simulation. The reporting stats show ~**4** cycle/req. But should DDR3 instead provide 1 cycle/req when the stream read/write accesses the same row? The verilog simulation with Micron DDR3 & Xilinx MIG shows that. 
The followings are my settings for ramulator simulation. Please let me know if there is something missing. Also, what is the difference between dram_cycles & ramulator_active_cycles in the reporting stats?

the config

`standard = DDR3
 channels = 1
 ranks = 1
 speed = DDR3_1600H
 org = DDR3_4Gb_x16
# record_cmd_trace: (default is off): on, off
 record_cmd_trace = off
# print_cmd_trace: (default is off): on, off
 print_cmd_trace = off

 cpu_tick = 4
 mem_tick = 1
 early_exit = on
 expected_limit_insts = 200000000
 warmup_insts = 0
 cache = no
 translation = None`

A part of the stream read trace is the following. It has a total of 9000 requests. The address interval is 8.

`0x0  R
0x8  R
0x10  R
0x18  R
0x20  R
0x28  R
0x30  R
0x38  R
0x40  R
0x48  R
0x50  R
0x58  R`

The report shows

> ramulator.dram_capacity          2147483648                                      # Number of bytes in simulated DRAM
> ramulator.dram_cycles               37159                                      # Number of DRAM cycles simulated
> ramulator.incoming_requests     9000                                      # Number of incoming requests to DRAM
> ramulator.read_requests                9000                                      # Number of incoming read requests to DRAM per core
>                                      [0]              9000.0                                      # 
> ramulator.write_requests                   0                                      # Number of incoming write requests to DRAM per core
>                                      [0]                 0.0                                      # 
> ramulator.ramulator_active_cycles               36108                                      # The total number of cycles that the DRAM part is active (serving R/W)

I print some information of Request in the read call back function, and got the followings

> addr=0, arrive@0, depart@23
> addr=8, arrive@1, depart@27
> addr=16, arrive@2, depart@31
> addr=24, arrive@3, depart@35
> addr=32, arrive@4, depart@39
> addr=40, arrive@5, depart@43
> addr=48, arrive@6, depart@47
> addr=56, arrive@7, depart@51
> addr=64, arrive@8, depart@55
> addr=72, arrive@9, depart@59
> addr=80, arrive@10, depart@63
> addr=88, arrive@11, depart@67
> addr=96, arrive@12, depart@71
> addr=104, arrive@13, depart@75
> addr=112, arrive@14, depart@79
> addr=120, arrive@15, depart@83
> addr=128, arrive@16, depart@87

olgunataberk, 2022-02-20:
Hi Miayiyi,

Why Ramulator reports four cycles: In DDR3 RD/WR requests are completed over 8 beats of a data transfer burst. The data is transmitted over the data bus, which operates at 2x the frequency of the command interface. Ramulator reports command interface clock cycles. Since four command bus cycles == eight data bus cycles, back to back READ requests to the same row takes four cycles in Ramulator.

Why MIG reports one cycle: The clock frequency of the Xilinx MIG is most likely 4x slower than the DRAM interface it controls. This is configurable to some extent based on the MIG version you are using and the FPGA device.

Miayiyi, 2022-06-08:
Thank you Ataberk! That perfectly answers my question.

closed 2022-06-08 by Miayiyi