CMU-SAFARI/ramulator #112: Problem with Ramulator.

zhizunbaowhy, 2022-07-04:
Hi,
To use ramulator.
I followed README.md

Memory trace Driven
$ cd ramulator
$ make -j
$ ./ramulator configs/DDR3-config.cfg --mode=dram dram.trace (here is problem)

ramulator: src/Config.cpp:43: void ramulator::Config::parse(const string &): Assertion `tokens.size() == 2 && "Only allow two tokens in one line"' failed.

I'm using the latest version. And I've already run "git pull". 
What should I do?
I would appreciate it if you could help me with the issue.

dohaisalive, 2023-04-15:
hi did you ever figure out how to solve the issue? i am getting the same error as well

XiongxiaoL, 2024-02-18:
Did you solve this issue?

XiongxiaoL, 2024-02-26:
Directly clone the project to Linux, some formats will change in Windows.