CMU-SAFARI/ramulator #115: Possible wrong address mapping

shenjiangqiu, 2022-09-05:
in file `src/Memory.h` 318-333

```c++
       else {
            switch(int(type)){
                case int(Type::ChRaBaRoCo):
                    for (int i = addr_bits.size() - 1; i >= 0; i--)
                        req.addr_vec[i] = slice_lower_bits(addr, addr_bits[i]);
                    break;
                case int(Type::RoBaRaCoCh):
                    req.addr_vec[0] = slice_lower_bits(addr, addr_bits[0]);
                    req.addr_vec[addr_bits.size() - 1] = slice_lower_bits(addr, addr_bits[addr_bits.size() - 1]);
                    for (int i = 1; i <= int(T::Level::Row); i++)
                        req.addr_vec[i] = slice_lower_bits(addr, addr_bits[i]);
                    break;
                default:
                    assert(false);
            }
        }
```

the code implicates two mapping schemes for those Config with a bank group like ddr4 or HBM. (`Bg` for bank group)

- the first case `ChRaBaRoCo` assumes `ChRaBgBaRoCo`.
- the second case `RoBaRaCoCh` assumes `RoBaBgRaCoCh`.( Should that be `RoBgBaRaCoCh` for the correct design?)
it that correct for the design?

If that is the case, I'll submit a pull request for it.

Dazhill, 2023-02-03:
I have the same problem.

Davidmicro, 2023-02-03:
I have the same problem.

joannahuadu, 2023-05-29:
Hi, recently I was also confused about this mapping, I wonder where you found the mapping scheme you proposed for ddr4 and HBM. All these mappings are black boxes?

In addition, I am very surprised that the "channel" is located in the low bit (e.g., RoBaBgRaCoCh). Does this mean that the addresses in a 4KB physical page will be assigned to different channels, that is, a 4KB physical page will store on different DDR DDIMs? Hereby ask for advice.

joannahuadu, 2023-05-29:
@shenjiangqiu  Hi, recently I was also confused about this mapping, I wonder where you found the mapping scheme you proposed for ddr4 and HBM. All these mappings are black boxes?

In addition, I am very surprised that the "channel" is located in the low bit (e.g., RoBaBgRaCoCh). Does this mean that the addresses in a 4KB physical page will be assigned to different channels, that is, a 4KB physical page will store on different DDR DDIMs? Hereby ask for advice.

shenjiangqiu, 2023-05-29:
> @shenjiangqiu Hi, recently I was also confused about this mapping, I wonder where you found the mapping scheme you proposed for ddr4 and HBM. All these mappings are black boxes?
>
> In addition, I am [...]

hi @joannahuadu , For HBM mapping scheme, you can find the code at src/HBM.cpp:12.

Yes, when Ch is at the lower bits, it assumes the BYTES are interleaved across the channels.

Furthermore, You can find my rewrite in rust: [ramu_rs](https://github.com/shenjiangqiu/ramu_rs)