CMU-SAFARI/ramulator #116: Error while using Trace Generator tool in Ramulator

Upasna55, 2022-11-05:
Hello All

I need to use trace generator tool in ramulator. Since, pin3.7 is not available now, so I've tried using pin versions `3.11, 3.22, and 3.25`, but I am getting the following error:
[image]
Quite possibly this seems to be a compiler compatibility issue. I am using `Ubuntu 20.04`, `Perl 5.30.0`, `gcc 9.4.0`. If you could help me in setting up trace generator, I would be so grateful.
 
Thanks & Regards
Upasna

ChenNie99, 2022-11-09:
Pintools 3.7 download link:

https://software.intel.com/sites/landingpage/pintool/downloads/pin-3.7-97619-g0d0c92f4f-gcc-linux.tar.gz

hope this helps you

Upasna55, 2022-11-09:
Yes Chen Nie, it worked. Thanks a lot. 
I need one more help. If you're using `vscode`, do you get `incomplete type` values of local variables while debugging Ramulator with `gdb`? If so, can you please share your configuration for appropriate debugging.  

Thanks & Regards
Upasna

Upasna55, 2022-11-13:
Yes Chen Nie, it worked. Thanks a lot.

Best Regards
Upasna

closed 2022-11-13 by Upasna55